bigtreetech/Pad7 #1: Backlight fully off?

Prutsium, 2023-05-01:
With the PAD7 the screen goes into sleep but the backlight stays on.
Is there a way to turn off the backlight when it's in sleep mode?

Yeltibe, 2023-06-04:
I would like to have this feature too.
Really annoying.

My crealtity sonic pad doesn't have this issue.

igeee81, 2023-06-05:
Hi,Look at this link. It was horribly for me, to find it in GitHub, last week. https://github.com/bigtreetech/CB1/releases/tag/V2.3.2Am 04.06.23 um 12:22 schrieb Yeltibe
                
            
            
                
                    Von: "Yeltibe" ***@***.***>Datum: 4. Juni 2023An: "bigtreetech/Pad7" ***@***.***>Cc: "Subscribed" ***@***.***>Betreff: Re: [bigtreetech/Pad7] Backlight fully off? (Issue #1)
                
                
I would like to have this feature too.
Really annoying.
My crealtity sonic pad doesn't have this issue.

royeiror, 2023-06-27:
I would love this as well

Abyss8, 2023-10-06:
Guys really? @bigtreetech 
[image]
[image]

Duudeeee, 2023-10-26:
anyone find a way to fix it? im on os 2.3.3 and still nothing

fernplant, 2023-11-12:
Yes, really a dumb oversight. Wish there was a fix

Baalsgate, 2026-10-16:
# BTT Pad 7: Turn off the hardware backlight when KlipperScreen blanks

I have a BIGTREETECH Pad 7 running KlipperScreen on a BTT-CB1. I found that when KlipperScreen blanked the display after its idle timeout, the LCD backlight remained on.

I implemented a small workaround that turns off the hardware backlight when KlipperScreen's screensaver activates and restores it when the screen wakes.

**Tested configuration**
- BTT Pad 7 with BTT-CB1
- BTT-CB1 2.3.2, Debian Bullseye
- Linux 5.16.17-sun50iw9
- KlipperScreen

Your PWM device and brightness-control process may differ if you have a different board or firmware.

## 1. Create the backlight helper

On this configuration, the backlight is controlled by `/sys/class/pwm/pwmchip0/pwm3`, while `/etc/scripts/auto_brightness` manages brightness automatically.

Create the helper:

```bash
sudo nano /usr/local/sbin/pad7-backlight
```

Add:

```bash
#!/bin/bash
set -e

PWM=/sys/class/pwm/pwmchip0/pwm3
STATE=/run/pad7-backlight-duty

case "$1" in
    off)
        PIDS=$(pgrep -x auto_brightness || true)

        if [ -n "$PIDS" ]; then
            kill -STOP $PIDS
        fi

        if [ ! -f "$STATE" ]; then
            cat "$PWM/duty_cycle" > "$STATE"
        fi

        echo 0 > "$PWM/duty_cycle"
        ;;

    on)
        if [ -f "$STATE" ]; then
            cat "$STATE" > "$PWM/duty_cycle"
            rm -f "$STATE"
        fi

        PIDS=$(pgrep -x auto_brightness || true)

        if [ -n "$PIDS" ]; then
            kill -CONT $PIDS
        fi
        ;;

    *)
        echo "Usage: pad7-backlight {off|on}" >&2
        exit 2
        ;;
esac
```

Save the file, then make it executable and root-owned:

```bash
sudo chown root:root /usr/local/sbin/pad7-backlight
sudo chmod 755 /usr/local/sbin/pad7-backlight
```

The helper saves the current PWM duty cycle before blanking the display, then restores it when the screen wakes. It pauses `auto_brightness` while the backlight is off so the background process does not immediately overwrite the zero duty cycle.

## 2. Test the helper manually

Run:

```bash
sudo /usr/local/sbin/pad7-backlight off
```

The backlight should turn off.

Then run:

```bash
sudo /usr/local/sbin/pad7-backlight on
```

The original brightness should return.

Test this manually before integrating it with KlipperScreen.

## 3. Permit KlipperScreen to invoke the helper

Create a dedicated sudoers file:

```bash
sudo visudo -f /etc/sudoers.d/pad7-backlight
```

Add this line:

```text
biqu ALL=(root) NOPASSWD: /usr/local/sbin/pad7-backlight off, /usr/local/sbin/pad7-backlight on
```

Here, `biqu` is the Linux account used by KlipperScreen on my installation. Substitute your actual account name if different.

Validate the configuration:

```bash
sudo visudo -cf /etc/sudoers.d/pad7-backlight
```

Check that both commands work without a password prompt:

```bash
sudo -n /usr/local/sbin/pad7-backlight off
sudo -n /usr/local/sbin/pad7-backlight on
```

## 4. Integrate with KlipperScreen

Back up the existing file first:

```bash
cp ~/KlipperScreen/ks_includes/widgets/screensaver.py \
   ~/KlipperScreen/ks_includes/widgets/screensaver.py.bak
```

In `ks_includes/widgets/screensaver.py`, add:

```python
import subprocess
```

In `ScreenSaver.show()`, after the existing call that turns off configured screen devices, add:

```python
subprocess.run(
    ["sudo", "-n", "/usr/local/sbin/pad7-backlight", "off"],
    check=False,
)
```

In `ScreenSaver.close()`, after the existing call that turns the screen devices back on, add:

```python
subprocess.run(
    ["sudo", "-n", "/usr/local/sbin/pad7-backlight", "on"],
    check=False,
)
```

Place these calls inside the corresponding methods, following the existing device on/off operations. Do not replace the existing screensaver logic.

Restart KlipperScreen using your normal service-management method.

**Note:** This modifies a KlipperScreen source file, so keep a backup. Updates may overwrite the change, and the patch should be checked against the version of KlipperScreen installed.

## 5. Verify the result

Leave the touchscreen idle until the configured timeout expires.

Expected behaviour:

- The LCD backlight turns off.
- Touching the screen wakes it.
- The backlight returns to its previous brightness.
- KlipperScreen continues to operate normally.

This worked on my Pad 7. Feedback from other Pad 7 and CB1 owners would be useful, particularly if they have different firmware versions or PWM configurations.