bigtreetech/Pad7 #3: Unable to set up input shaping

jeffstark, 2023-05-19:
I've gone over the documentation numerous times now and I'm pretty sure everything is configured correctly. When I enter the input shaping section on the pad 7, this is what the console tells me:

`1:32 PM
Unknown command:"MEASURE_AXES_NOISE"
1:32 PM
accelerometer values (x, y, z): 9193.145976, 370.102971, -0.000000
1:32 PM
shaper_type_y:mzv shaper_freq_y:38.200 damping_ratio_y:0.100000
1:32 PM
shaper_type_x:ei shaper_freq_x:79.600 damping_ratio_x:0.100000`

Since it is reporting those values, it would seem that the accelerometer is functional and accessible to the pad. I'm able to see the accelerometer in ifconfig and I can grep it. 

The green LED on the board is on.

My printer.cfg has the proper entries to enable the module:

`[mcu CB1]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: CB1:None
spi_bus: spidev1.1
axes_map: z,y,-x`

I'm just unable to run the input shaping calibration.

I've tried changing the switch on the pad to cm4 as I've seen suggested here and on discord, but that doesn't help.

Any ideas?

Vash2081, 2023-05-26:
Hi Jeff, looks like I am having the same issue with my pad 7, where the input shaping button on the klipper screen is not showing up.

my configuration on my printer cfg looks like this. please let me know what I am missing. thank very much.

[mcu rpi]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: rpi:None
spi_bus: spidev1.1
axes_map: z,y,-x

[adxl345]
cs_pin: rpi:None
spi_speed: 2000000
spi_bus: spidev1.1

[resonance_tester]
accel_chip: adxl345
accel_per_hz: 70
probe_points:
    147, 154,10

jeffstark, 2023-05-26:
Some things I would try based on your config would be:

- combine the two adxl345 sections into one and remove the duplicate values (i have no idea if this is an issue or not, it could very well be perfectly fine the way it is)
- your probe points seem to be missing a space before the last value (again, this could be perfectly fine, it just stuck out to me as looking odd)

I just looked over the [user manual](https://github.com/bigtreetech/Pad7/blob/master/BIGTREETECH%20Pad%207%20v1.0%20User%20Manual.pdf) and there are a couple things for you to check. If you are using a raspberry pi like your config seems to indicate, you need to flip the switch on the pad7 to the CM4 position (page 10 in the doc). Also, the spi_bus for the pi is spidev0.1 (as opposed to spidev1.1 for the CB1. Page 22 in the doc.).

I'm no expert, I just barely found enough information to make this functional on my end. Hopefully someone that knows what they are talking about can chime in.

Here is my config for reference, but I am using a CB1, and it appears you are using a raspberry pi based on your config

`[mcu CB1]
serial: /tmp/klipper_host_mcu

[adxl345]
cs_pin: CB1:None
spi_bus: spidev1.1
axes_map: z,y,-x

[resonance_tester]
accel_chip: adxl345
probe_points: 100, 100, 20`

I hope this helps!

Vash2081, 2023-05-26:
Hey Jeff! thank you very much for your response, I was able to figure it out, basically I had to follow the Klipper documentation using the CLI to perform the resonance tests ( using the commands TEST_RESONANCES AXIS = X & Y). Once the test were done I was finally able to see input shaping on the Klipper screen.

so I am good to go now lol thanks again.

closed 2023-05-26 by jeffstark