bigtreetech/Pad7 #4: WIFI not working - bad antenna design.

alfrankgit, 2023-06-02:
Yesterday I got my BTT Pad 7 an tried for hours to get it to connect to my WIFI properly and stay connected for more than a few seconds: no chance!

As I've been an IT service technician since 1988 I tried everything: new OS image, all kinds of authentication and WIFI generation settings, connecting via nmtui in the linux shell. Nothing brought me any closer to a stable connection, so I reset everything to the state it was before which works for all my other, very diverse, equipment. The Pad 7 still failed to stay connected.

Today I took a simple 2.4 GHz WIFI antenna with me from work, removed the heat sink from the CB1 module, unplugged the internal WIFI antenna, plugged mine in, put the heat sink back on, turned the Pad 7 on: it instantly connected to my WIFI router.

And it doesn't disconnect anymore and has good ping times.

Now I got curious and removed the back panel of the Pad 7 (caution: remove µSD card first), disconnected the speakers and took a look at the printed foil circuit antenna. 

I couldn't believe what I saw: the antenna trace (the "snake in the lower horizontal part of the foil) is connected directly to the GND surface (big vertical rectangle in the top part), see the first picture.

To be sure I measured the resistance between the shielding and the core of the antenna cable: 0 ohms, see second picture.

ARE YOU SERIOUS???

Please revise your antenna design ASAP or your customer service will be flooded with WIFI problems!

[image]

[image]

alfrankgit, 2023-06-02:
Quick and dirty hack for the more daring users:

Cut the antenna trace directly adjacent to the GND plate.

[image]

Gixie69, 2023-06-26:
Tried connecting the Pad7 i received today, Trys twice and gives up then freezes the unit. Have to turn off and on to restart. Im going to have to hard wire and that wasnt what i needed.

alfrankgit, 2023-06-26:
> Tried connecting the Pad7 i received today, Trys twice and gives up then freezes the unit. Have to turn off and on to restart. Im going to have to hard wire and that wasnt what i needed.

You have to replace or "fix" the antenna as I did. It's a hardware issue: a very badly designed antenna.

alfrankgit, 2023-06-26:
[image]

This is the antenna that ships with the standalone CB1 boards. It looks like a much better design but I haven't used it yet.
At least there's no connection between the ground and the antenna traces.

Gixie69, 2023-06-26:
I have two of the CB1 antennas, Ones for the Manta M8P i can hopefully get going now, The others for the cb1/pi4b adapter im going to use with the Mini e3v3 thats currently running marlin. I take it they dont have the little button socket to install the better antenna

Gixie69, 2023-06-26:
Swapped the antenna and still nothing. Im wondering if its the 2.4ghz Pad 7 to my 5ghz router thats the problem here.

Gixie69, 2023-06-26:
[image]

GavinHanson, 2023-07-19:
> Quick and dirty hack for the more daring users:
> 
> Cut the antenna trace directly adjacent to the GND plate.
> 
> [image]
> 
> This results in a working WIFI antenna which still has at least double the ping times of my external one.

Did this on my one, it seems to pick up loads of wifi networks now but connects to none of them!

Gixie69, 2023-08-15:
I still havent gotten mine working right. The main connection trouble was SKY HUB, Even set up for 2.4ghz. I brought a new router which runs just 2.4ghz and the Pad connected straight away and updated everything without any trouble. BUT, And heres the next nail in its coffin, Putty cant pick it up even with the IP address the Pad 7s showing. Its just not there. So i tryed  WinSCP, Angry IP scanner too, Nothing.  so whats wrong and why are BTT not coming back to me on the support tickets?  They did once which i replied to with as much information as i could put together but they just blanked me.  Im guessing theres a problem with the CB1 itself so its being replaced with a Raspberry CM4 module as soon as there back in stock.  They are tested before their sent out, Something BTT should start doing by the looks of things.

GavinHanson, 2023-08-15:
I ended up removing the entire wifi antenna from mine, drilling a small hole and fitting one of these, its been flawless since, also on Sky Hub.

https://www.amazon.co.uk/Bingfu-Antenna-Wireless-Express-Network/dp/B0B5KY4KV3/ref=sr_1_12?crid=2304E8ZZV81F7&keywords=mini+Wifi+antenna&qid=1692102641&sprefix=mini+wifi+antenna%2Caps%2C94&sr=8-12

[image]
[image]
[image]

Check the BTT Pad 7 Facebook page, there's a few pics of my mod on there and quite a few people willing to help. BTT arent known for quick responses.

Gixie69, 2023-08-15:
Cool, Good plan.  Mine connects now, its just got no detectable IP for putty to find. Ive litrally just tried it again and very nearly smashed it to bits because the darn thing wouldn’t turn on. I since realised what was wrong there, The laptop I just put a new Putty on has a battery, The Pad 7 dont. Id turned off the extension lead last night after a dream inspired by a fellow printer about fires with wrong firmware. Ffs!!! Will this nightmare never end lol.

Very luckierly ive taken to walking away first, good job too of id be posting pictures of one very broken Pad 7.

Im now having a coffee and ill be back to it.

Still, That don’t explain why Everything ive brought direct from BTT don’t work.   The Mini e3v3 they sent was flashed for a CR10 bedslinger which a Ender 5 plus it is not. Zero BTT firmware for the E5+ you have to go by TH3D and use their firmware, A modified Marlin. I spent weeks trying to get it working right, But that final press print after all the levelings worked, everythings heated up right, The bed drops and it prnts in thin air, Ill run to the end of the print too if you let it.
The Manta m8p boards still in the box, If the Mini e3 wont work im not depressing myself trying the manta.

I replaced them both wwith MKS Nano v1.2 and v3,1, There both turning out to be a arse ache too. I think because the printers have been moved about so much the steppers are out of wack somehow and im not 100% sure how to correct that, hopefully if I get the Pad7 working....  But then the Nanos aint Klipperised yet so...

Aint this a funny old world.

https://www.amazon.co.uk/Bingfu-Antenna-Wireless-Express-Network/dp/B0B5KY4KV3/ref=sr_1_12?crid=2304E8ZZV81F7&keywords=mini+Wifi+antenna&qid=1692102641&sprefix=mini+wifi+antenna%2Caps%2C94&sr=8-12

[8B126A88-E05F-4C0C-860B-52C440443232_1_105_c]<https://user-images.githubusercontent.com/3464631/260710056-e53915bf-37e9-445c-a41d-535dda594918.jpeg>
[0A12586E-E0E3-405E-A884-129DEA62A0BC_1_105_c]<https://user-images.githubusercontent.com/3464631/260710154-c76f9ffa-20d7-4b35-b03c-7931fad8dfab.jpeg>
[6DADA795-EAD3-42FA-9024-5ED38B3F78E4_1_105_c]<https://user-images.githubusercontent.com/3464631/260710232-6c2c07a7-221f-45e5-a895-0ac269f5e09e.jpeg>

GavinHanson, 2023-08-15:
Haha I'll take that manta off your hands for my voron 😂

Hugowhitee, 2024-04-02:
2024 and Im having the same problems. Antenna has a very bad design. Does someone know if I can connect it with ethernet to my wifi booster? I always get a failed message.