bigtreetech/Pad7 #7: ADXL345 does not work with Klipper

ah-banny, 2023-07-28:
the Pad 7 had the original klipper OS on the SD card,  version with CB1,
[image]

configure Pad7 in BoardEnv.txt
[image]

now configure in klipper
[image]
[image]

control by putty
[image]

restart Klipper, it works, 
[image]

but when I run the command "ACCELEROMETER_QUERY" in the clipper console I get this error:
Invalid adxl345 id (got ff vs e5).
[image]

what is the problem?

thanks for help :-)

BahamutWolf, 2023-08-25:
I have the same problem and have not been able to find a solution. Maybe a shielded LAN cable can help. I haven't tried it yet but I will try it soon. With me USB Choose right and CS Choose left. When CS Choose is set to right on the CB1, the error message changes every time I enter the command ACCELEROMETER_QUERY.

Andy3dprinter, 2023-10-02:
The whole process is very bad and extremely difficult. It took me several days to get it to work and I then tried to get 2 instances on the pad 7 so i can run 2 printers and now i can't even install 2 instances without the pad crashing and not booting back up and i also lost input shaper. I had copied my CFG file and when I have the exact same  thing however now my ADXL345 no longer works. And I get an MCU error. It makes no  sense as this input shaper does not have a MCU