bigtreetech/Pad7 #8: How to install multiple instances?

Andy3dprinter, 2023-10-02:
I have tried and tried to install multiple instances and cannot get it to work. I use KIAUH and once i delete klipper , moonraker, mainsail, and crowsnest it will not allow me to add klipper back on again. I got it to work once and then I was able to reinstall everything back on and when i rebooted the pad 7 it would no longer boot back up again. There seems to be so many issues and I am starting to think it is related to the last few updates with klipper , moonraker etc. I also tried reflashing Cb1 OS on the pad 7 and used V2.3.2 and this is also not letting me add more than 1 instance. Anyone else having these issues?

Gixie69, 2023-10-03:
Ive had a Pad7 since it first came out and the CB1 in an adapter, Neither have ever worked. When i find some Raspi 4b's in stock im going to change the CB1 out and see if i have luck with Klipper running through the Pi4b. If not itll go in the bin as im sick of looking at My big expensive pile of BTT component's that I in the UK have had zero joy with.   My ender 5 plus is in the shop with a Mini e3v3 i could do nothing with, Their also having issues so its not just my Lorry drivers brain messing things up. My Pad 7 connects to the BB/net and everythings updated except the actual Klipper screen so it wont connect to my PC or Printers

sweenbeen, 2024-04-24:
I followed the following video from Aurora Tech
[https://youtu.be/8vkzpvia4mg?si=Dzqe_z1CGyn5jG_f](url)

My new issue/question is - how do you go about changing the printer - say you get rid of it or swap it out with a new printer - how do you go about updating, or do you just do a fresh reinstall?