bigtreetech/Pad7 #18: Cannot get EBB42 to connect via CANBUS

Aryeh56, 2025-03-20:
Hello, I am having a very difficult time getting the CAN network started on my PAD 7. I have no idea if the issue is with the PAD 7 or the EBB. I have taken the following steps to configure the system so far:

- Created a CANBUS klipper image for the EBB42 and flashed it to the EBB42 using STM32CubeProgrammer. However, neither Klipper nor Katapult have an option in the "make" menu to use "PB0/PB1". Both of them have "PD0/PD1" options instead that I used as my best guess. Screenshot attached.

[image]
 
- Enabled mcp2515 in the Pad7's BoardEnv.txt
- Verified the CAN network is running using ifconfig, lsusb, etc.
- Reviewed the /networkd configurations and confirmed they are as specified
- Verified the wires are ordered correctly and getting good connection. Photo attached.

[image]

- Added a jumper to the "120R" pins on the EBB42

Despite all of this, I am getting no device UUIDs on the CAN network, nor any traffic whatsoever. Please advise on any other diagnostic steps I should try.

The following network checks were done with the same configuration pictured above:

```
biqu@BTT-Pad7:~$ ifconfig
can0: flags=193<UP,RUNNING,NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1024  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether 56:53:b5:21:f6:f4  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 12620 (12.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 34

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 901  bytes 445173 (434.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 901  bytes 445173 (434.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
```
biqu@BTT-Pad7:~$ ~/klippy-env/bin/python ~/klipper/scripts/canbus_query.py can0

Total 0 uuids found
biqu@BTT-Pad7:~$
biqu@BTT-Pad7:~$
```

Aryeh56, 2025-03-23:
Hello again - I fixed the issue with not having "PB0". That part was my simple mistake. I had the wrong processor selected. However, I have now re-flashed the device with firmware that exactly matches BTT's specifications and I am still having the problem.

Aryeh56, 2025-03-23:
Another update for posterity - I did find an issue with the BoardEnv.txt file. The pad I had shipped with the ws2812 overlay enabled. The formatting Biqu used for this line also enables both the mcp2515 and spidev1_1 overlays, for CANBUS and SPI respectively. This was causing a pin conflict between the CAN and SPI systems which was visible when running the `dmesg | grep can` command.

Unfortunately, I did not think to get a screenshot of the issue. However, I was able to fix it by deactivating spidev in the BoardEnv file.

Meanwhile, my cat yanked on the cable I was using to check the EBB42 at my desk... and managed to drop the board to the floor with enough force to break off the connections for the capacitor. I am going to trade the cat for a fresh board and continuing to work this issue once I have it. Any further feedback from BTT/Biqu or anyone else who has worked on CANBUS is still welcome in the meantime.

bigtreetech, 2025-03-23:
We have had issues with catbus before too. We found that if you pay too much attention to it then it stops working but if you look away and ignore it then it craves attention and becomes more compliant. Toolboards and keyboards are common places for catbus to perch so you want to keep them out the way.

We will check into your issue with the env file first thing tomorrow ;)

Aryeh56, 2025-03-25:
Hello again. I see you and I have a similar model of "catbus" lol.

I have found the final error in my system, at least the final one related to this topic. The can0 network was set to a bit-rate of 500,000. I was using 1,000,000 for the firmware as recommended by the [EBB42 manual](https://github.com/bigtreetech/EBB/blob/master/EBB%20CAN%20V1.1%20and%20V1.2%20(STM32G0B1)/EBB42%20CAN%20V1.1%20and%20V1.2/BIGTREETECH%20EBB42%20CAN%20V1.1%20User%20Manual.pdf) on page 21.

I fixed this by changing the bitrate on the Pad's network to 1000000. However, I could've also fixed it by building the firmware at 500000. It only occurred to me to check because I was reading [this guide](https://canbus.esoterical.online/troubleshooting/no_uuid.html). I would recommend to Biqu that either adjust future pad models default settings or revise the manual for the EBB42. I am now showing the EBB42 in Klipper and am closing my issue accordingly.

closed 2025-03-25 by Aryeh56